dmslabsbr/hoymiles #137: Support for hybrid inverters from hoymiles

hangingloose, 2024-02-19:
Hello, 

are the hybrid inverters (HAS / HYS / HAT / HYT) tested / supported? 
If not, is there a timeline for this inverters?

I hope to get my HYS this week and it would be great, if I could implement this inverter also  ;-)
I will use the inverter only for a battery. 

it would be great if i could get access to the timeperiode where the inverter should load the battery from grind!
Then I can user a dynamic power supplier like tibber. 
With tibber I get the price for the power in 1h steps one day bevore.

Cosik, 2024-02-20:
@hangingloose sorry, I hadn't access to such installation, maybe it will work. I tried to do this as an semi dynamic addon. When you will have this installation please let me know and we will see what is possible to be done.

hangingloose, 2024-02-29:
How can I send you a private message? I didn’t find this option…
I get the Information, that the inverter + storage is now on stock 😉
The only question is the installation…

Cosik, 2024-03-12:
@hangingloose only contact over discord is possible, My id is Cosik#6329 then we could discuss what is missing and not working.

hangingloose, 2024-03-26:
Hello Cosic, 
my electrician will come in 2 weeks.
Tried to send you a Discord invite. (hopefully) 
Then I need some information so that I can set you up as a plant user.

Cosik, 2024-04-19:
#144 implement this setup with some restrictions which could be found in readme file.

hangingloose, 2024-04-19:
thank you for your support!

closed 2024-04-19 by hangingloose