dmslabsbr/hoymiles #138: DTU with status  Connect put not connect...

focabr, 2024-02-21:
I hope all is well with you and your family.

Excellent work you are doing on this project, congratulations to everyone!

I installed and configured the edge version today, it works as it should.

However, I went to do a test after 2 hours and it was already working normally and I disconnected the DTU, as shown in the images below it continues to state that it is connected, but I unplugged it, and on the s-Miles cloud website the information is correct saying that it is turned off.

Follow the evidence.

[image]

[image]

[image]

In your code there is a condition that evaluates whether it is connected, I will put the link to your code and the payload for your evaluation.

https://github.com/dmslabsbr/hoymiles/blob/master/edge/hoymilesapi.py#L373

POST: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree

```JSON
{"status":"0","message":"success","data":[{"id":1430210,"sn":"10D881824282","vc":"","dtu_sn":"10D881824282","type":1,"version":3,"replace_num":0,"model_no":"DTU-Lite-S","soft_ver":"V00.03.07","hard_ver":"H06.02.01","extend_data":{},"warn_data":{"_rw":"1","connect":false,"warn":false},"children":[{"id":6554090,"sn":"116483907751","vc":"","dtu_sn":"10D881824282","type":3,"version":3,"replace_num":0,"model_no":"HMS-2000-4T","soft_ver":"V01.00.18","hard_ver":"H00.04.00","extend_data":{"grid_name":"","port_array":[1,2,3,4],"grid_id":0,"grid_version":""},"warn_data":{"warn":false,"connect":false},"children":[]},{"id":6554080,"sn":"116483906889","vc":"","dtu_sn":"10D881824282","type":3,"version":3,"replace_num":0,"model_no":"HMS-2000-4T","soft_ver":"V01.00.18","hard_ver":"H00.04.00","extend_data":{"grid_name":"","port_array":[1,2,3,4],"grid_id":0,"grid_version":""},"warn_data":{"warn":false,"connect":false},"children":[]},{"id":6554110,"sn":"116483905769","vc":"","dtu_sn":"10D881824282","type":3,"version":3,"replace_num":0,"model_no":"HMS-2000-4T","soft_ver":"V01.00.18","hard_ver":"H00.04.00","extend_data":{"grid_name":"","port_array":[1,2,3,4],"grid_id":0,"grid_version":""},"warn_data":{"warn":false,"connect":false},"children":[]},{"id":6554100,"sn":"116483905413","vc":"","dtu_sn":"10D881824282","type":3,"version":3,"replace_num":0,"model_no":"HMS-2000-4T","soft_ver":"V01.00.18","hard_ver":"H00.04.00","extend_data":{"grid_name":"","port_array":[1,2,3,4],"grid_id":0,"grid_version":""},"warn_data":{"warn":false,"connect":false},"children":[]}]}],"systemNotice":null}
```

Cosik, 2024-02-22:
@focabr addon is focused on pooling mainly PV data from HM cloud. PV data is refreshed ~15min. Did you check status after 15-20min? I could reduce pooling time but it will generate more traffic to cloud. 

To be honest now I know how I could rewrite whole addon to change behavior and flow but I don't know if it is worth to do (I'm no longer using this plugin - just try to maintain it)

Cosik, 2024-02-22:
please check #139 reduce poling time to 8min and sending mqtt to 5 min. Is it ok, or should be less? In general decreasing mqtt time will cause that DB will growth with duplicated data.

focabr, 2024-02-23:
Thank you very much for your feedback and clarifications.

Using the new version, it still has the same problem showing "connected".

I did the following, I turned off the DTU for 1h10 on the website https://global.hoymiles.com/platform/station/view/detail/device?id=1535490 it appeared offline and also all the Micros, but on homeassistant it always appeared "connected".

[image]

One of the things that caught my attention was this condition where the answer is the same.

https://github.com/dmslabsbr/hoymiles/blob/master/edge/hoymilesapi.py#L68

Could this be related?

focabr, 2024-02-25:
Hello, after many hours of DEBUG I found what the problem was related to this issue.

To be able to use DEBUG I first had to fix it, which wasn't working at least for me, I found the problem and submitted a PR.

After much analysis I discovered that it was a string to integer conversion. I made the correction and sent it or PR.

Please feel free to contact me!

closed 2024-03-16 by focabr