dmslabsbr/hoymiles #145: HYT: Integrate temperature values

richie241, 2024-04-25:
I use the Edge version and the connection to my Hoymiles HYT + battery works well. Thank You.

Is it possible to integrate the temperature values of the inverter + fan speed? I would like to build an external fan control for the device if it gets too warm.

[image]

Cosik, 2024-04-25:
@richie241 there is an info on main page about addon. You have a good idea but bad concept of realization.

1. Cloud has 15min refresh rate.
2. Using cloud as a temperature source for additional cooling is bad idea because it will not work without internet.
3. You have device on your house hold locally so why you don't check device temperature directly?
4. I propose to use thermo control fans and no internet nor other communication is need.

richie241, 2024-06-16:
I have previously measured the external temperature, but I would like to measure the internal temperature and use it to control my ventilation BEFORE the device gets warm and the internal fan starts.

Can you please release the parameters?

Cosik, 2024-06-16:
Which parameters? I have no access to installation which provide mentioned by you parameters, so it is not possible from my side to do it.

richie241, 2024-06-16:
all temperature values.
the data should be transferred with an HYT.

how can i query the api, then i will send you the result.

Cosik, 2024-06-16:
I haven't any HYT. You have to use traffic analysis in developer tool in your web browser.

richie241, 2024-06-17:
I found it, maybe the 4 parameters can be integrated into your query:

name: "Internal fan speed", val: "0", unit: "RPM", ...
key "inv_spd1"

name: "Inverter heat sink temperature", val: "45", unit: "℃", ... }
key "inv_tinv"

name: "PV heat sink temperature", val: "41", unit: "℃", ... }
key "inv_tpv"

name: "Internal ambient temperature", val: "47", unit: "℃", ... }
key "inv_tin"

Cosik, 2024-06-17:
I don't know which query provide such parameters, what is more I need full response to parse it. Also I put information in Readme that extending functionality with present implementation is very hard.