dmslabsbr/hoymiles #147: brauche hilfe

stefan190977, 2024-05-09:
2024-05-09 12:14:20,415 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.3.0
2024-05-09 12:14:20,416 - HoymilesAdd-on - INFO - Starting up... 2024-05-09 12:14:20
2024-05-09 12:14:20,417 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2024-05-09 12:14:20,418 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2024-05-09 12:14:20,419 - HoymilesAdd-on.mqttapi.Mqtt - INFO - mqtt.Client e69b30d0-0dec-11ef-a854-d83addda9f4d
2024-05-09 12:14:20,419 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT core-mosquitto
2024-05-09 12:14:20,424 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2024-05-09 12:14:20,425 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to core-mosquitto
2024-05-09 12:14:21,426 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2024-05-09 12:14:21,766 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2024-05-09 12:14:21,768 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2024-05-09 12:14:22,120 - HoymilesAdd-on - ERROR - User has no access to plant
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Cosik, 2024-05-09:
Podejrzewam iż zostały wpisane niepoprawne dane.

stefan190977, 2024-05-09:
S-Miles-Cloud daten habe ich richtig ein gegeben

stefan190977, 2024-05-09:
[16:56:37] INFO: /data/secrets.ini not exists!
Getting mqtt data...
[16:56:38] INFO: null
[16:56:39] INFO: {"supervisor":"2024.05.1","homeassistant":"2024.5.2","hassos":"12.1","docker":"24.0.7","hostname":"homeassistant","operating_system":"Home Assistant OS 12.1","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"raspberrypi4-64","arch":"aarch64","state":"running","supported_arch":["aarch64","armv7","armhf"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Vienna"}
[16:56:39] INFO: {"supervisor":"2024.05.1","homeassistant":"2024.5.2","hassos":"12.1","docker":"24.0.7","hostname":"homeassistant","operating_system":"Home Assistant OS 12.1","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"raspberrypi4-64","arch":"aarch64","state":"running","supported_arch":["aarch64","armv7","armhf"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Vienna"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2024-05-09 16:56:39,686 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.1.6
2024-05-09 16:56:39,687 - HoymilesAdd-on - INFO - Starting up... 2024-05-09 16:56:39
2024-05-09 16:56:39,687 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2024-05-09 16:56:39,688 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2024-05-09 16:56:39,688 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2024-05-09 16:56:40,563 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2024-05-09 16:56:40,564 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2024-05-09 16:56:41,589 - HoymilesAdd-on - ERROR - User has no access to plant
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

Cosik, 2024-05-10:
No ale to prawie taki sam log. Wrzucanie tego samego nie powoduje zwiększenia ilości informacji w konsekwencji czego ktokolwiek byłby w stanie pomóc.

stefan190977, 2024-05-12:
It has lost the Plant ID.
Plant ID is the number of my balcony plant, which is displayed in the address bar of the browser when I go to the picture of my plant on the Hoymiles page mentioned above

Cosik, 2024-05-12:
So, is it working now or not? If still not working I need debug logs from your addon.

stefan190977, 2024-05-12:
It works now without any problems, the only thing missing was the Panet ID.

Thank you

closed 2024-05-12 by Cosik