dmslabsbr/hoymiles #148: Could it be possible to get 2 more Grid information

Wind06, 2024-05-11:
Hello,

I would like to know if 2 other data can be collected:

- To  and From Grid

This will be very helpfull for building statistics with HomeAssistant.
Below is a screen shot. Of course I will be please to help in case more information or testing requiered.

Thanks.

[image]

Cosik, 2024-05-12:
@Wind06 but those values are already provided by edge version

Wind06, 2024-05-12:
you are right, meter informations are povided. To avoid local calculation I was wondering if the "to Grid" and "From Grid" can be downloaded from the cloud ?

Cosik, 2024-05-12:
But to grid and from grid are not reading from meter?

Wind06, 2024-05-12:
what is read from grid is negative or positive value depending if From or To grid.
To Grid is positive value
From Grid is negative value
Value are reported in "sensor.hoymiles_gateway_solarh_775030_meter_grid_power"

[image]

Cosik, 2024-05-12:
@Wind06 but there are "meter_to_grid" and "meter_from_grid" which are already provided and those are energy values. Those values could be used in HA energy dashboard and it will calculate all details.

Wind06, 2024-05-12:
indeed yes, but the meter_from_grid is always at 0W and the from_grid do have an impressive value which is not the value I have in the hoymiles dashboard.

Cosik, 2024-05-12:
Ok, so as I mentioned, I will need access.

Cosik, 2024-05-14:
I checked and found that probably there is a bug in cloud data getting. Addon is checking total data. On checked installation total data of return to grid value is 0. When I checked month by month return values to grid are non zero. In present implementation it is not possible to fix it. It will required changing type of meter data from constant increasing to something other. Also request has to be changed. 

Maybe in new release it will be possible.

closed 2024-05-14 by Cosik