dmslabsbr/hoymiles #149: HoyMiles Solar Gateway Stable Addon - Installation failed

VolkerLaude, 2024-06-05:
Home Assistant

Core
2024.5.5
Supervisor
2024.05.1
Operating System
12.3
Frontend
20240501.1

Installation ends with this error message: The command '/bin/ash -o pipefail -c pip3 install paho-mqtt==1.6.1' returned a non-zero code: 1

Home assistant log:
2024-06-05 11:42:10.202 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /addons/44349f66_hoymiles_solardata/install - The command '/bin/ash -o pipefail -c pip3 install paho-mqtt==1.6.1' returned a non-zero code: 1

rocketrobin777, 2024-08-12:
I also get the same error:
The command '/bin/ash -o pipefail -c pip3 install paho-mqtt==1.6.1' returned a non-zero code: 1

Core: 2024.8.1
Supervisor: 2024.08.0
OS: 12.4

Cosik, 2024-08-12:
Hi #173 proof that installation of addon is working. I need more details to fully reproduce issue.

rocketrobin777, 2024-08-12:
> Hi, #173 proof that installation of addon is working. I need more details to fully reproduce issue.

Hi,

after updating HA to latest version, I tried to install the stable addon by using repository link.
Click on install starts install process, but it fails during installation.

[image]

Cosik, 2024-08-12:
To be honest I just checked instalation on my HA test instance on VBox and it was working correctly

`[20:50:31] INFO: {"supervisor":"2024.08.0","homeassistant":"2024.8.1","hassos":"11.5","docker":"24.0.7","hostname":"homeassistant","operating_system":"Home Assistant OS 11.5","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"}
[20:50:31] INFO: {"supervisor":"2024.08.0","homeassistant":"2024.8.1","hassos":"11.5","docker":"24.0.7","hostname":"homeassistant","operating_system":"Home Assistant OS 11.5","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","haos","resolved","journal","disk","mount"],"machine":"qemux86-64","arch":"amd64","state":"running","supported_arch":["amd64","i386"],"supported":true,"channel":"stable","logging":"info","timezone":"Europe/Warsaw"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2024-08-12 20:50:31,328 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.4.1`

rocketrobin777, 2024-08-12:
ok, thx. I can't believe, why it crashes in my case

Cosik, 2024-08-13:
@rocketrobin777 me neither. Are there maybe some additional info in logs? Or did you tried old stable?

rocketrobin777, 2024-08-13:
@Cosik by trying to install the old stable version, I get a hint:
[image]
"Add on cannot be installed"

After trying to install the stable version again, I get the same error, but had a vier to the logs:

Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:135
Integration: Home Assistant Supervisor (Dokumentation, Probleme)
First log: 10:18:32 (1 Vorkommnisse)
Last log: 10:18:32

Failed to to call /addons/44349f66_hoymiles_solardata_stable/install - The command '/bin/ash -o pipefail -c pip3 install paho-mqtt==1.6.1' returned a non-zero code: 1

Cosik, 2024-08-13:
Last idea which you could check is try to use my fork with bumped container image, please use this link as installation source https://github.com/Cosik/hoymiles

rocketrobin777, 2024-08-13:
@Cosik running with the forked repo works fine on my raspi4. Thank you very much.
You'll find the log from my first successful request below

2024-08-13 17:32:34,082 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.4.0
2024-08-13 17:32:34,082 - HoymilesAdd-on - INFO - Starting up... 2024-08-13 17:32:34
2024-08-13 17:32:34,084 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2024-08-13 17:32:34,085 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2024-08-13 17:32:34,085 - HoymilesAdd-on.mqttapi.Mqtt - INFO - mqtt.Client 42f975b6-5989-11ef-87be-dca6324d64c4
2024-08-13 17:32:34,085 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT core-mosquitto
2024-08-13 17:32:36,977 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2024-08-13 17:32:36,978 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to core-mosquitto
2024-08-13 17:32:37,453 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2024-08-13 17:32:40,980 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2024-08-13 17:32:40,981 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2024-08-13 17:32:41,992 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2024-08-13 17:32:42,716 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2024-08-13 17:32:43,448 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2024-08-13 17:32:43,874 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2024-08-13 17:32:44,267 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2024-08-13T17:20:22+02:00
2024-08-13 17:32:44,267 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2024-08-13 17:32:44,656 - HoymilesAdd-on - INFO - Solar data publication...2024-08-13 17:32:44.655852
2024-08-13 17:32:44,658 - HoymilesAdd-on - INFO - DTU-WLite_1432940 data publication...2024-08-13 17:32:44.657947
2024-08-13 17:32:44,659 - HoymilesAdd-on - INFO - HM-1500_6604660 data publication...2024-08-13 17:32:44.659441
2024-08-13 17:32:44,665 - HoymilesAdd-on - INFO - Main loop start!

closed 2025-07-17 by Cosik