dmslabsbr/hoymiles #152: 502 Bad Gateway Error Breaks Fetching

veilofsecurity, 2024-06-20:
I have been getting a 502 error pretty regularly now and when it occurs the add-on stops fetching new information until a restart. A restart will fix for a short period but then a 502 returns and breaks it again.

Probably just need to catch the 502 and handle gracefully to continue the loop and retry after a couple minutes.
 
2024-06-20 02:03:13,417 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Status code: 502Bad Gateway
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/hoymiles.py", line 302, in run
    self.execute(*self.args, **self.kwargs)
  File "/hoymiles.py", line 236, in publicate_data
    hoymiles_h.get_alarms()
  File "/hoymilesapi.py", line 402, in get_alarms
    retv = self.send_payload(DATA_FIND_DETAILS, header, payload)
  File "/hoymilesapi.py", line 354, in send_payload
    retv = json.loads(retv)
  File "/usr/local/lib/python3.9/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.9/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/lib/python3.9/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

Cosik, 2024-06-21:
@veilofsecurity in general which version you are using? I checked code and it seems that you are using some very old implementation.

First of all, try use latest one and then update ticket.

veilofsecurity, 2024-06-21:
I was using "HoyMiles Solar Gateway Addon" version 1.0.11 as I had auto updates on, saw no updates available, saw no notes in the changelog, and no indication that things were moved to a different addon requiring reinstallation. I have moved to "HoyMiles Solar Gateway Stable Addon" now, thanks. 

Will update if I see the 502 issue again.

closed 2024-06-21 by veilofsecurity