dmslabsbr/hoymiles #154: Incorrect values on meter_from_grid and meter_to_grid

xezpeleta, 2024-07-06:
Hi!

There is an issue with the meter values:

In Home Assistant I see the entities "meter_from_grid" and "meter_to_grid" updated once a day, when they should be updated every 15 min.

The JSON retrieved from the API using hoymiles.py is different from the Hoymiles app.

[image]

If we use Curl or the browser inspector, the JSON contains the correct values (which are updated every 15 min):

[image]

Cosik, 2024-07-06:
@xezpeleta you are wrong. Your are showing present values. Addon is providing values form begging so they are constant increasing type

xezpeleta, 2024-07-08:
Hi @Cosik 

Maybe I'm wrong, but If that's true, shouldn't be both values updating every 15 min?

At least in my computer they are updating every 24h. So we can't use them for the Home Assistant Energy dashboard.

That's why I've started looking for the source of the issue, and I've discovered the JSON retrieved by the addon was different from the one loaded in my browser. I've changed the addon and I'm happily using the patched version.

If you prefer, we can wait until someone else confirm the issue.

Thanks!

Cosik, 2024-07-08:
Dashboard on HM cloud are updating values every 15 minutes. So far request which were used was to get total value, not value from present day. Your PR changing request to totally different one.

Cosik, 2024-07-08:
They are correct. They should be marked as constant increasing so after about 24h energy dashboard in HA should count normally differences and show proper consumption. This dashboard was designed to use these values, I'm using it and it is calculated ok. Also people with whom I was adding support config proper calculation.

xezpeleta, 2024-07-08:
I understand, so then you are getting "meter_to_grid" and "meter_from_grid" lefetime data on purpose.

In my opinion the main handicap is you can't see the evolution during the day and you get a value which is totally different from the HM cloud UI "to grid" and "from grid" info ( #148 )

Maybe the best solution to avoid breaking changes is to add two new entities. But probably nobody is asking this except me
.. so forget it 😅

You can close the issue and the PR. Thanks for your job!

Cosik, 2024-07-12:
Sorry I don't understand. Using this values you are able to see it. What is more you could add manual custom counter of energy which is reset every day. I also have it.

About extending addon there is information on Readme about situation.

Cosik, 2024-07-12:
Ok, now I have access to PC and remember that ther is a bug in HM cloud database. It returns wrong values in total request. But in present implementation fixing it is not so easy as it is in Readme