dmslabsbr/hoymiles #156: TypeError: float() argument must be a string or a number, not 'NoneType'

guidoffm, 2024-07-11:
```
b'{"status":"0","message":"success","data":{"today_eq":"4000","month_eq":"0","year_eq":null,"total_eq":"0","real_power":"3775.0","co2_emission_reduction":"0","plant_tree":"0","data_time":"2024-07-11 16:38:50","last_data_time":"2024-07-11 16:38:50","capacitor":"13","is_balance":0,"is_reflux":1,"reflux_station_data":{"start_date":"2024-07-11","end_date":"2024-07-11","pv_power":"3775.0","grid_power":"-3240.0","load_power":"7015.0","bms_power":"0.0","bms_soc":"0","inv_num":1,"meter_location":2,"pv_to_load_eq":"4000","load_from_pv_eq":"4000","meter_b_in_eq":"3500","meter_b_out_eq":"0","bms_in_eq":"0","bms_out_eq":"0","self_eq":"0","pv_eq_total":"4000","use_eq_total":"7400","flows":[{"out":2,"in":1,"v":0},{"out":4,"in":1,"v":0}],"icon_pv":1,"icon_grid":1,"icon_load":1,"icon_bms":0,"icon_gen":0,"icon_pvi":0,"mb_in_eq":{"today_eq":"3500","month_eq":"0","year_eq":"0","total_eq":"0"},"mb_out_eq":{"today_eq":"0","month_eq":"0","year_eq":"0","total_eq":"0"},"menu_pv":0,"pile_power":"0"}},"systemNotice":null}'
2024-07-11 16:39:47,506 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - solar_data: {'today_eq': '4000', 'month_eq': '0', 'year_eq': None, 'total_eq': '0', 'real_power': '3775.0', 'co2_emission_reduction': '0', 'plant_tree': '0', 'data_time': '2024-07-11 16:38:50', 'last_data_time': '2024-07-11 16:38:50', 'capacitor': '13', 'is_balance': 0, 'is_reflux': 1, 'reflux_station_data': {'start_date': '2024-07-11', 'end_date': '2024-07-11', 'pv_power': '3775.0', 'grid_power': '-3240.0', 'load_power': '7015.0', 'bms_power': '0.0', 'bms_soc': '0', 'inv_num': 1, 'meter_location': 2, 'pv_to_load_eq': '4000', 'load_from_pv_eq': '4000', 'meter_b_in_eq': '3500', 'meter_b_out_eq': '0', 'bms_in_eq': '0', 'bms_out_eq': '0', 'self_eq': '0', 'pv_eq_total': '4000', 'use_eq_total': '7400', 'flows': [{'out': 2, 'in': 1, 'v': 0}, {'out': 4, 'in': 1, 'v': 0}], 'icon_pv': 1, 'icon_grid': 1, 'icon_load': 1, 'icon_bms': 0, 'icon_gen': 0, 'icon_pvi': 0, 'mb_in_eq': {'today_eq': '3500', 'month_eq': '0', 'year_eq': '0', 'total_eq': '0'}, 'mb_out_eq': {'today_eq': '0', 'month_eq': '0', 'year_eq': '0', 'total_eq': '0'}, 'menu_pv': 0, 'pile_power': '0'}}
Traceback (most recent call last):
  File "/hoymiles.py", line 429, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/hoymiles.py", line 385, in main
    publicate_data(plant_list[id], mqtt_list[id])
  File "/hoymiles.py", line 238, in publicate_data
    hoymiles_h.get_solar_data()
  File "/hoymilesapi.py", line 237, in get_solar_data
    self.solar_data = self.adjust_solar_data(self.solar_data)
  File "/hoymilesapi.py", line 277, in adjust_solar_data
    round(float(solar_data['year_eq']) / 1000, 2))
TypeError: float() argument must be a string or a number, not 'NoneType'
```

Cosik, 2024-07-11:
What I can say in this case, look at closed issues. This was discussed few times.

guidoffm, 2024-07-11:
> What I can say in this case, look at closed issues. This was discussed few times.

Any link?

Cosik, 2024-07-11:
Nope, please use search. This should be the first thing before creating new issue.

closed 2024-07-21 by Cosik