dmslabsbr/hoymiles #159: Download Production History

Alex-Jud, 2024-07-19:
Hello, I'm asking here, because you seems to be the most advanced project using the S-Miles interface.

I'm currently working with the Unofficial Hoymiles API and attempting to fetch historical production data for each panel from the endpoint :
https://neapi.hoymiles.com/pvm-data/api/0/station/data/count_playback_power_by_day

I'm able to log in, get the valide token, and have a 200 response.
However, the response I'm receiving is quite garbled. Here’s an example of the response string:
```
POST Response: ??????????[HQ?#
{?|??z	?=s?]?,v?t_B??b??)??)?????????]oi??]-b??R??????H?DS?o?O/A
????}?????KNKu?SYz?Z?+???`4d?F?????HK?? 3#????M??27en?<?y(?P?,??L?2)?t2F??`?8)'??c0R????F?'?d?&#?I9??HqRNF?`0R???Q/?za??^?¨F?0?Q/?zaT?B¨F???Lk???x???$?????|????6??$V?U?
?Q?H?????Z9?*?rR?&kT+?O?1?:?$??M??.?????W?????LXG5a?k?PH?3B??\??Fsp?]X?u?_??sI??x??????G?T?sN?R?8?jn??jU?+R9WTi? ??Hz?b???[???y??_???{gl?'?m????s'qyW???????????`Smxa?????Mk??'??%P?l??Fx
4?M??G@?????h^?	?????@?K???h?)n??"O~??@G??iP???<
???v?W??^?C??/O??????gFA?}???(h?[?G? ??@????>?!??C?Y???a??g@c???u?'K????????Gf@???~(??e??aP??
e?w???????~??0?W?yu??z??Og}?o??:????
```
The response seems to be in an unreadable format, and I'm unsure if this is due to encoding issues or if there's a specific way to decode this data.

Have you encountered this before, or does you know how to properly decode this response? Any advice or suggestions on how to handle this would be greatly appreciated!

Thanks in advance for your help!

Cosik, 2024-07-21:
This is packed data. There was some tries to get more details for cloud but effort is not worth it. Add has to be redesign. As you notice this is reverse engineering, they could provide API doc but they request some money for it.

Alex-Jud, 2024-07-21:
I also tried with https://global.hoymiles.com/platform/api/gateway/pvm-report/report_export_station_data to retrieve a report in CSV format.

In the browser, I get the following response:
```
{
    "status": "0",
    "message": "success",
    "data": {
        "file_name": "station_power_data_2024-07-20.zip",
        "file_path": "tmp/9876ca7565804a71a586f3c0d4a77775.zip",
        "url": "https://dl.hoymiles.com/tmp/9876ca7565804a71a7777a586f3c0d45.zip?AccessKeyId=TSYOGTLJZFOFZNTQGKHC&Expires=1721557619&Signature=kRW2nA4nllf6yi0bxUoAQ8ApNLT%3D"
    },
    "systemNotice": null
}
```

But in the application, I get a response:
`{"status":"200" ...... ` when sending.
However, I receive:
`{"status":"100","data":"","message":"header authorization not null.","systemNotice":null}`
when reading the content.

The request, header, and authorizations seem to match those in the browser...

In short, it's all dead ends with reverse engineering this site...

Cosik, 2024-07-21:
Did you used latest version? Edge or stable? Did you ping url from PC on which addon is working?

dpobel, 2024-11-01:
Hello, I'm currently trying to reverse engineer those APIs as well. For now, not everything is 100% crystal clear here are my findings so far in case somebody else gets here (sorry for somehow hijacking this issue :) )

- https://neapi.hoymiles.com/pvm-data/api/0/station/data/count_playback_power_by_day (and probably all others endpoints of neapi.hoymiles.com) seems to use protobuf as a serialization mechanism. And to deserialize the response, a _protobuf message_ is needed (some kind of schema) that is yet to be figured out (the webapp JS code seems to hold one but I haven't yet really checked)
- @Alex-Jud dl.hoymiles.com and global.hoymiles.com seem to require the token to be in a cookie called `hm_token` so that the requests work.

Hope this helps