dmslabsbr/hoymiles #160: Addon fails / freezes in an udatected by HA manner.

jgrobelny-dev, 2024-07-21:
My wild guess is that during some internet outages, or HA updated I get into a weird place when HA thinks that addon is running, while it's not. Here is the log for this event. Manual restart solves the issue, watchdog doesn't help :(
I can't replicate the error but perhaps some logs would help:

```
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/hoymiles.py", line 302, in run
    self.execute(*self.args, **self.kwargs)
  File "/hoymiles.py", line 234, in publicate_data
    hoymiles_h.update_devices_status()
  File "/hoymilesapi.py", line 299, in update_devices_status
    status, hws_data = self.request_plant_hw()
  File "/hoymilesapi.py", line 337, in request_plant_hw
    retv = self.send_payload(GET_ALL_DEVICE_API, header, payload)
  File "/hoymilesapi.py", line 352, in send_payload
    retv, s_code = self.send_request(
  File "/hoymilesapi.py", line 180, in send_request
    response = sess.send(prepped)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 700, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='global.hoymiles.com', port=443): Max retries exceeded with url: /platform/api/gateway/pvm/station_select_device_of_tree (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x7ff1a1d44730>: Failed to resolve 'global.hoymiles.com' ([Errno -3] Try again)"))
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
Exception ignored in: <module 'threading' from '/usr/local/lib/python3.9/threading.py'>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 1470, in _shutdown
    lock.acquire()
  File "/hoymiles.py", line 279, in signal_handler
    raise ProgramKilled
__main__.ProgramKilled: 

```During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/hoymiles.py", line 302, in run
    self.execute(*self.args, **self.kwargs)
  File "/hoymiles.py", line 234, in publicate_data
    hoymiles_h.update_devices_status()
  File "/hoymilesapi.py", line 299, in update_devices_status
    status, hws_data = self.request_plant_hw()
  File "/hoymilesapi.py", line 337, in request_plant_hw
    retv = self.send_payload(GET_ALL_DEVICE_API, header, payload)
  File "/hoymilesapi.py", line 352, in send_payload
    retv, s_code = self.send_request(
  File "/hoymilesapi.py", line 180, in send_request
    response = sess.send(prepped)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 700, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='global.hoymiles.com', port=443): Max retries exceeded with url: /platform/api/gateway/pvm/station_select_device_of_tree (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x7ff1a1d44730>: Failed to resolve 'global.hoymiles.com' ([Errno -3] Try again)"))
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
Exception ignored in: <module 'threading' from '/usr/local/lib/python3.9/threading.py'>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/threading.py", line 1470, in _shutdown
    lock.acquire()
  File "/hoymiles.py", line 279, in signal_handler
    raise ProgramKilled
__main__.ProgramKilled:

Cosik, 2024-07-21:
Hi,

You not provided all details. What is more add-on show that you have issue with connection to HM cloud, so it is not an addon issue

jgrobelny-dev, 2024-07-21:
Sure, I see that. 
The log report just looked to me like an unhandled exception of some sort in hoymilesapi.py. I'm not sure about precise versions of HA at the moment error occured, as it was at the brink of my HA update.
Now as I have updated all the HA stuff (Core 2024.7.3 Supervisor 2024.06.2 Operating System 12.4 Frontend 20240710.0)
 and Hoymiles AddOn  (now 1.0.11) I will provide all the details if it occurs again.  My internet connection becomes unstable during harsh weather conditions (LTE), nevertheless lack of internet access should not affect long term data gathering from the addon. While this error make the addon come to a stop and doesn't allow HA watchdog to restart it. Manual addon restart solves the issue.