dmslabsbr/hoymiles #162: Power Limiter for Hoymiles

matohle, 2024-07-30:
Hey, 

I´m a bit new to Home Assistant but I would like to know if I can limit the output power with your add on. 

I have already installed the plug in and I get the correct values from my S-Miles DTU a -> Hoymiles. 

But I would like to reduce or increase the power due to some automazations. Is this possible?

As well it seems the communication intervals are very long (update of the values every 15-20min). Can we reduce this to a few seconds?

Thanks,

Matthias

Cosik, 2024-07-31:
@matohle no possibility for reduce power is not implemented. When new design of addon will be done there will be room for new features.

About the time. What will give you if we reduce time interval? There are many reasons to not do that.

matohle, 2024-07-31:
Hey @Cosik ,

thanks for the fast Feedback. 

Setting lower the interval would only make sense when I can control the power of the Hoymiles. I saw videos were people are doing this with openDTU and AhoyDTU and I thought it should work with the DTU Lite / Pro with a Hoymiles too, because it’s possible in the S-Miles Cloud. 

I want to realize an automatic limiter when my consumption is lower than the output from my batteries. 

I use 2 Anker batteries on my solar system with a Hoymiles 1400. 

I will check what I can do instead. 

Thanks for your help. Much appreciate.

Matthias

Cosik, 2024-07-31:
But you mentioned solutions which are communicating directly with inverters. This addon is communicating with cloud and data on cloud has 15min refresh rate, so polling data more often is without sense.

matohle, 2024-07-31:
Make sense. I’m new into this topics and it makes absolutely sense what you are saying. I will check for a DTU which is able to communicate without a cloud. Thanks!

closed 2024-08-13 by Cosik