dmslabsbr/hoymiles #168: TypeError

NikolasK-source, 2024-08-03:
2024-08-03 20:56:33,128 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2024-08-03 20:56:34,464 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2024-08-03 20:56:34,464 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2024-08-03 20:56:34,831 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2024-08-03 20:56:35,176 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2024-08-03 20:56:35,535 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2024-08-03 20:56:35,887 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2024-08-03 20:56:36,237 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - real_power = 0
Traceback (most recent call last):
  File "/hoymiles.py", line 499, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/hoymiles.py", line 454, in main
    publicate_data(plant_list[id], mqtt_h)
  File "/hoymiles.py", line 263, in publicate_data
    hoymiles_h.get_solar_data()
  File "/hoymilesapi.py", line 265, in get_solar_data
    self.solar_data = self.adjust_solar_data(self.solar_data)
  File "/hoymilesapi.py", line 308, in adjust_solar_data
    last_data_time = datetime.strptime(last_data_time, "%Y-%m-%d %H:%M:%S")
TypeError: strptime() argument 1 must be str, not None

Benthegreat, 2024-08-04:
I have the same error message.

Cosik, 2024-08-04:
To start working on this issue you have to provide more information.

NikolasK-source, 2024-08-06:
What information do you need?

Cosik, 2024-08-06:
Version and type of addon. Full log with error with enabled debug logging and developer mode.

NikolasK-source, 2025-01-25:
I could not replicate the issue, maybe it is fixed