dmslabsbr/hoymiles #173: Add-on HoyMiles Solar Gateway Edge Addon has been remove

Kwiatjk, 2024-08-07:
Add-on HoyMiles Solar Gateway Edge Addon has been removed from the repository it was installed from. This means it will not get updates, and backups may not be restored correctly as the supervisor may not be able to build/download the resources required.

Clicking submit will uninstall this deprecated add-on. Alternatively, you can check [Home Assistant help](https://www.home-assistant.io/help/) and the [community forum](https://community.home-assistant.io/) for alternatives to migrate to.

Cosik, 2024-08-07:
[image]
[image]

Kwiatjk, 2024-08-07:
If i set that version will bi problems with encje ?
i lost statatistic etc ?

Cosik, 2024-08-07:
If you delete them from db you will lost it. Edge was just released to stable...

abpei, 2024-08-07:
> If i set that version will bi problems with encje ? i lost statatistic etc ?

I stopped and disabled the **Edge** addon, installed the **Stable** addon, added Hoymiles login credentials and started it. It publishes the same entities to the MQTT server so nothing else needs to be reconfigured and everything else keeps working. At least that was the case for me. 

Thanks for all your effort here, Cosik, it is appreciated.

Kwiatjk, 2024-08-08:
Thx is working

closed 2024-08-12 by Cosik