dmslabsbr/hoymiles #175: ERROR User has no acces to plant

a4tje, 2024-08-12:
hello, I have got a strange issue, everything was working fin until suddenly it stopped, without me making any changes, i got the following message in the log: HoymilesAdd-on - ERROR - User has no access to plant.
I tried 1111 or 11111 as hoymiles plant id but still got this message.
Here is the log:
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2024-08-12 20:28:17,097 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.4.1
2024-08-12 20:28:17,097 - HoymilesAdd-on - INFO - Starting up... 2024-08-12 20:28:17
2024-08-12 20:28:17,098 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2024-08-12 20:28:17,099 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2024-08-12 20:28:17,099 - HoymilesAdd-on.mqttapi.Mqtt - INFO - mqtt.Client a4b0147e-58d8-11ef-8d7c-02dc2847f9ec
2024-08-12 20:28:17,099 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT core-mosquitto
2024-08-12 20:28:17,104 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2024-08-12 20:28:17,105 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to core-mosquitto
2024-08-12 20:28:18,106 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2024-08-12 20:28:18,796 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2024-08-12 20:28:18,797 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_find
2024-08-12 20:28:19,675 - HoymilesAdd-on - ERROR - User has no access to plant
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

any clues would be nice!

Cosik, 2024-08-12:
You have to provide correct plant id. I'm sure that 1111 or 11111 are not your plant ID...

a4tje, 2024-08-13:
yes, thought of dat ofcourse, to my knowledge it should be 1111, but i still get the same error .
[image]

Cosik, 2024-08-13:
Nope, it is not plant ID. If you don't know how to get it, it is always in url

a4tje, 2024-08-13:
Ok, thanks for your reply, Cosik,
I'm not sure what you mean by your answer, it is alway in url? witch url?
it's in this one?
BlueMail <https://bluemail.me> for Desktop downloaden
Cosik wrote:
Nope, it is not plant ID. If you don't know how to get it, it is always in url
—

Dit e-mailbericht is met Avast-antivirussoftware gecontroleerd op virussen.
www.avast.com

Cosik, 2024-08-13:
In general plant ID is located on hoymiles web, but easier is to take it from the URL when you open your plant dashboard on hoymiles web

a4tje, 2024-08-13:
I've worked it out myself:
- log in to your smiles cloud
- go to settings, the monkey wrench in the lower left corner of the screen
- then in the new screen on your right, push settings
- the last numbers after id= in your url is the plant-id

closed 2024-08-13 by a4tje