dmslabsbr/hoymiles #177: Can not see / find all entities for each device / inverter

MiniPV, 2024-09-02:
In MQTT I can see 12 Devices with 58 entities.

For each inverter I can see only following sensors:

solarH_xxxxxxxxxx_alarm_code
solarH_xxxxxxxxxx_alarm_string
solarH_xxxxxxxxxx_connect

This was working in the past.

Best regards,

Wolfgang from MiniPV.at
[image]
[image]

MiniPV, 2024-09-02:
from attached screenshot you can see all my Hoymiles devices and from plant device I do not get details from each inverter.
[image]
[image]

Cosik, 2024-09-02:
Because you will not get data from each inverter. In easy way each inverter is providing only some information about status and alarm. This addon didn't provide data from each inverter only from plant...