dmslabsbr/hoymiles #178: Can not install Edge version in Home Assistant

MiniPV, 2024-09-02:
I'm very sorry - but how can I install upcoming (developer) Edge Version?

When searching in Add-on Store in can only find:

HoyMiles Solar Gateway oldStable Addon (can not be installed 44349f66_hoymiles_solardata_oldStable)

and

HoyMiles Solar Gateway stable Addon
where version 1.4.1. is installed but not listing all entities (still opened this issue)

That's why I would try it with Edge version - but sorry I'm not able to do it.

Best regards,

Wolfgang from [MiniPV.at]
[image]

Cosik, 2024-09-02:
Edge was released recently to stable. Edge is rewritten from scratch

MiniPV, 2024-09-02:
sorry - confused ......

So then I'm still using Edge when installing stable version today?

Best regards,

Wolfgang from MiniPV.at

closed 2024-09-02 by MiniPV

Cosik, 2024-09-02:
No, time to time when edge is mature enough it should be released and called stable...