dmslabsbr/hoymiles #180: Fails when there is no data yet

augustynski-lukasz, 2024-09-20:
Addon failes on newly installed PV plant when there is no data yet.

Logs:
2024-09-20 15:25:11,945 - HoymilesAdd-on.hoymilesapi.Hoymiles - WARNING - real_power = 0
Traceback (most recent call last):
  File "/hoymiles.py", line 499, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/hoymiles.py", line 454, in main
    publicate_data(plant_list[id], mqtt_h)
  File "/hoymiles.py", line 263, in publicate_data
    hoymiles_h.get_solar_data()
  File "/hoymilesapi.py", line 265, in get_solar_data
    self.solar_data = self.adjust_solar_data(self.solar_data)
  File "/hoymilesapi.py", line 308, in adjust_solar_data
    last_data_time = datetime.strptime(last_data_time, "%Y-%m-%d %H:%M:%S")
TypeError: strptime() argument 1 must be str, not None

xxLeoxx93, 2024-10-26:
Same here a month later…

Cosik, 2024-10-26:
@xxLeoxx93 but it is only for fresh installation. where there is no data on the server. It will be fixed in new version.