dmslabsbr/hoymiles #182: No new production data since couple of days. An issue?

Krzysztonek, 2024-09-29:
My configuration has not been changed since moths, the Add-on worked quite fine for me so far.
Moreover, I don't see any errors in logs. Hoymiles devices work nice, the power production is being registered by my home power meter.  

However, since Sept 25th 2024 the add-on does NOT collect any new daily production data:
[image]
On HA dashboard my daily production is displayed as '0', which is not true.

Is it an issue or the Hoymiles add-on is currently working in some maintenance mode, and will come back soon to the working mode?

Cosik, 2024-09-30:
On your installation it look strange, you have 'None_95151' instead of inverter type. Also adding debug log and info about used version will help. Based on what you put in ticket I can say that edge version is down.

markbeee, 2024-09-30:
Mine also didn't collect any new data. I had to start the add-on again. Maybe check for this.

Krzysztonek, 2024-09-30:
> I had to start the add-on again

Either restart or re-installation of add-on from scratch (full uninstall, then install again) doesn't help in my case.

Krzysztonek, 2024-10-01:
> Also adding debug log and info about used version will help.

This is my debug log (already sanitized, where 'xxxxx' is my station id):
```
2024-10-01 12:46:15,903 - HoymilesAdd-on - DEBUG - 'switch'
2024-10-01 12:46:15,903 - HoymilesAdd-on - DEBUG - 'number'
2024-10-01 12:46:15,903 - HoymilesAdd-on - DEBUG - .../solarH_95151_alarm_code
2024-10-01 12:46:15,903 - HoymilesAdd-on - DEBUG - {"stat_t": "home/solar/json_95151", "name": "solarH_95151_alarm_code", "uniq_id": "solarH_95151_alarm_code", "val_tpl": "{{ value_json.alarm_code }}", "icon": "mdi:alarm-light-outline", "expire_after": "720", "device": {"name": "Hoymiles Gateway", "manufacturer": "Hoymiles", "model": "None", "sw_version": "V00.01.38", "via_device": "95151", "identifiers": ["solarH_95151"]}}
2024-10-01 12:46:15,904 - HoymilesAdd-on - DEBUG - .../solarH_95151_alarm_string
2024-10-01 12:46:15,904 - HoymilesAdd-on - DEBUG - {"stat_t": "home/solar/json_95151", "name": "solarH_95151_alarm_string", "uniq_id": "solarH_95151_alarm_string", "val_tpl": "{{ value_json.alarm_string }}", "icon": "mdi:alarm-light-outline", "expire_after": "720", "device": {"name": "Hoymiles Gateway", "manufacturer": "Hoymiles", "model": "None", "sw_version": "V00.01.38", "via_device": "95151", "identifiers": ["solarH_95151"]}}
2024-10-01 12:46:15,904 - HoymilesAdd-on - DEBUG - .../solarH_95151_connect
2024-10-01 12:46:15,904 - HoymilesAdd-on - DEBUG - {"stat_t": "home/solar/json_95151", "name": "solarH_95151_connect", "uniq_id": "solarH_95151_connect", "val_tpl": "{{ value_json.connect }}", "icon": "mdi:solar-panel", "device_class": "connectivity", "device": {"name": "Hoymiles Gateway", "manufacturer": "Hoymiles", "model": "None", "sw_version": "V00.01.38", "via_device": "95151", "identifiers": ["solarH_95151"]}}
2024-10-01 12:46:15,904 - HoymilesAdd-on - DEBUG - Hass Sended for micro_95151
2024-10-01 12:46:15,905 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2024-10-01 12:46:15,905 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - payload: 
{
    "body": {
        "id": xxxxx
    },
    "WAITING_PROMISE": true
}

2024-10-01 12:46:15,905 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - {'Content-Type': 'application/json;charset=UTF-8', 'Cache-Control': 'no-cache', 'Host': 'global.hoymiles.com', 'Connection': 'keep-alive', 'Accept': 'application/json, text/plain, */*', 'Accept-Encoding': 'gzip, deflate, br', 'User-Agent': 'Mozilla/5.0 (Macintosh; Intel Mac OS X 11_2_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.90 Safari/537.36', 'Accept-Language': 'pt-BR,pt;q=0.9,it-IT;q=0.8,it;q=0.7,es-ES;q=0.6,es;q=0.5,en-US;q=0.4,en;q=0.3', 'Cookie': "'uid=fff9c382-389f-4a47-8dc9-c5486fc3d9f5; hm_token=2.1zcJfdyOU0rVhjCnH70TZIG1tdWyJQCtoGhhTZKMuqoGwSFi1ZXmxnnOpCCh47sDNLpTuynCGr8y.0; Path=/; Domain=.global.hoymiles.com;Expires=Sat, 30 Mar 2025 22:11:48 GMT;'", 'Content-Length': '67'}
2024-10-01 12:46:16,446 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - content: b'{"status":"0","message":"success","data":[{"id":12122,"sn":"10F052403007","vc":"","dtu_sn":"10F052403007","type":1,"version":2,"replace_num":0,"model_no":"DTU-MI","soft_ver":"V10.01","hard_ver":"V04.00","extend_data":{},"warn_data":{"_rw":"1","connect":false,"warn":false},"children":[{"id":95154,"sn":"104151301132","vc":"","dtu_sn":"10F052403007","type":3,"version":2,"replace_num":0,"model_no":null,"soft_ver":"V00.02.36","hard_ver":"H13.00.01","extend_data":{"grid_name":"","port_array":[1,2],"grid_id":0,"grid_version":""},"warn_data":{"warn":false,"connect":false},"children":[]},{"id":95153,"sn":"106151004388","vc":"","dtu_sn":"10F052403007","type":3,"version":2,"replace_num":0,"model_no":null,"soft_ver":"V00.01.38","hard_ver":"H04.00.00","extend_data":{"grid_name":"","port_array":[1,2,3,4],"grid_id":0,"grid_version":""},"warn_data":{"warn":false,"connect":false},"children":[]},{"id":95152,"sn":"106151005815","vc":"","dtu_sn":"10F052403007","type":3,"version":2,"replace_num":0,"model_no":null,"soft_ver":"V00.01.38","hard_ver":"H04.00.00","extend_data":{"grid_name":"","port_array":[1,2,3,4],"grid_id":0,"grid_version":""},"warn_data":{"warn":false,"connect":false},"children":[]},{"id":95151,"sn":"106151004197","vc":"","dtu_sn":"10F052403007","type":3,"version":2,"replace_num":0,"model_no":null,"soft_ver":"V00.01.38","hard_ver":"H04.00.00","extend_data":{"grid_name":"","port_array":[1,2,3,4],"grid_id":0,"grid_version":""},"warn_data":{"warn":false,"connect":false},"children":[]}]}],"systemNotice":null}'
2024-10-01 12:46:16,448 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2024-10-01 12:46:16,448 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - payload: 
{
    "body": {
        "sid": xxxxx
    },
    "WAITING_PROMISE": true
}

2024-10-01 12:46:16,450 - HoymilesAdd-on.hoymilesapi.Hoymiles - DEBUG - {'Content-Type': 'application/json;charset=UTF-8', 'Cache-Control': 'no-cache', 'Host': 'global.hoymiles.com', 'Connection': 'keep-alive', 'Accept': 'application/json, text/plain, */*', 'Accept-Encoding': 'gzip, deflate, br', 'User-Agent': 'Mozilla/5.0 (Macintosh; Intel Mac OS X 11_2_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.90 Safari/537.36', 'Accept-Language': 'pt-BR,pt;q=0.9,it-IT;q=0.8,it;q=0.7,es-ES;q=0.6,es;q=0.5,en-US;q=0.4,en;q=0.3', 'Cookie': "'uid=fff9c382-389f-4a47-8dc9-c5486fc3d9f5; hm_token=2.1zcJfdy [... cut ...]
2024-10-01 12:46:20,093 - HoymilesAdd-on - INFO - None_95153 data publication...2024-10-01 12:46:20.093895
2024-10-01 12:46:20,095 - HoymilesAdd-on - INFO - None_95152 data publication...2024-10-01 12:46:20.095552
2024-10-01 12:46:20,097 - HoymilesAdd-on - INFO - None_95151 data publication...2024-10-01 12:46:20.097151
2024-10-01 12:46:20,099 - HoymilesAdd-on - INFO - Main loop start!
```

But, what is worse, the official Hoymiles Monitoring website also does NOT show the power production since 26.09.2024:

[image]

Cosik, 2024-10-01:
So if you don't see data on HM page that mean they has some issues or your DTU is not sending data. Add-on has nothing with this issue

Krzysztonek, 2024-10-01:
> they has some issues or your DTU is not sending data

This is what I have suspected...

Anyway, thanks a lot for your support!