dmslabsbr/hoymiles #184: MqttSensor is using state class 'measurement'

jimjaeger, 2024-10-20:
HoyMiles Solar Gateway stable Addon
Current version: 1.4.1

HMS-800W-2T

Open a ticket because warning following warning in log:
2024-10-19 01:16:48.344 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.hoymiles_gateway_solarh_<removed>_real_power_measurement (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

Cosik, 2024-10-26:
@jimjaeger it is not an error, only warning. What solution you propose?

jimjaeger, 2025-04-29:
The message say when it is energy change from measurement to total, or total_increasing.

Cosik, 2025-04-29:
@jimjaeger solution is to remove it, do you agree?

jimjaeger, 2025-05-06:
I agree - the info is available in 2 other sensors. but I am not sure who depends on.