dmslabsbr/hoymiles #185: Cant connect on hoymiles

vfernandes0408, 2024-11-01:
I had a dtu-w100 that data is send to web.
When i try to connect Addon i receive some errors

`drwxr-xr-x    2 root     root          4096 Nov  1 13:22 templates
[13:23:04] INFO: /data/secrets.ini not exists!
Getting mqtt data...
[13:23:05] ERROR: Got unexpected response from the API: Service not enabled
[13:23:06] INFO: null
[13:23:06] INFO: {"supervisor":"2024.10.3","homeassistant":"2024.10.4","hassos":null,"docker":"27.3.1","hostname":"orangepizero3","operating_system":"Debian GNU/Linux 12 (bookworm)","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","resolved","journal","disk","mount"],"machine":"odroid-n2","arch":"aarch64","state":"running","supported_arch":["aarch64","armv7","armhf"],"supported":true,"channel":"stable","logging":"info","timezone":"America/Sao_Paulo"}
[13:23:06] INFO: {"supervisor":"2024.10.3","homeassistant":"2024.10.4","hassos":null,"docker":"27.3.1","hostname":"orangepizero3","operating_system":"Debian GNU/Linux 12 (bookworm)","features":["reboot","shutdown","services","network","hostname","timedate","os_agent","resolved","journal","disk","mount"],"machine":"odroid-n2","arch":"aarch64","state":"running","supported_arch":["aarch64","armv7","armhf"],"supported":true,"channel":"stable","logging":"info","timezone":"America/Sao_Paulo"}
dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2024-11-01 13:23:07,657 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.4.1
2024-11-01 13:23:07,658 - HoymilesAdd-on - INFO - Starting up... 2024-11-01 13:23:07
2024-11-01 13:23:07,660 - HoymilesAdd-on - INFO - Using Internal MQTT Server: 
2024-11-01 13:23:07,660 - HoymilesAdd-on - INFO - Using Internal MQTT User: 
2024-11-01 13:23:07,661 - HoymilesAdd-on.mqttapi.Mqtt - INFO - mqtt.Client 942c526e-986d-11ef-967b-aca4c8530a01
2024-11-01 13:23:07,661 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT 
2024-11-01 13:23:07,661 - HoymilesAdd-on.mqttapi.Mqtt - DEBUG - SSL: OpenSSL 1.1.1o  3 May 2022
2024-11-01 13:23:07,662 - HoymilesAdd-on.mqttapi.Mqtt - DEBUG - mqttStart TLS: False
2024-11-01 13:23:07,662 - HoymilesAdd-on.mqttapi.Mqtt - DEBUG - mqttStart MQTT_USERNAME: 
2024-11-01 13:23:07,662 - HoymilesAdd-on.mqttapi.Mqtt - DEBUG - mqttStart MQTT_PASSWORD: 
2024-11-01 13:23:07,663 - HoymilesAdd-on.mqttapi.Mqtt - DEBUG - mqttStart MQTT_PORT: 1883
2024-11-01 13:23:07,663 - HoymilesAdd-on.mqttapi.Mqtt - ERROR - Invalid host.`

Cosik, 2024-11-01:
Wrong configured addon as log shows
`HoymilesAdd-on.mqttapi.Mqtt - ERROR - Invalid host.`

vfernandes0408, 2024-11-01:
I only configured user, pass and plant id.
Mqtt is need ?
[image]

Cosik, 2024-11-01:
it required connection to mqtt broker for proper working.

vfernandes0408, 2024-11-01:
> it required connection to mqtt broker for proper working.

but mqqt local or online ? because my solar energy system is not local

Cosik, 2024-11-01:
you can use any mqtt broker unless it is connected to your home assistant.

vfernandes0408, 2024-11-01:
> you can use any mqtt broker unless it is connected to your home assistant.
Now, i guess that connected
[image]

closed 2025-05-01 by Cosik