dmslabsbr/hoymiles #188: Adjust update time

BastiBerlin93, 2025-03-17:
Hello, is it possible to set the update time shorter ? for example 30s or something like that

Cosik, 2025-04-08:
@BastiBerlin93 yes but cloud is updating data every 15 min so 30s will overload cloud. This info is in documentation and other tickets

BastiBerlin93, 2025-04-08:
@Cosik The cloud has been updating every 5 minutes since the last update, so it would be great if the add-on could also do this

Cosik, 2025-04-08:
@BastiBerlin93 if data is not changed so often what will be benefit? If I good remember now it is 5-8 minutes.

BastiBerlin93, 2025-04-08:
@Cosik I hoped that the real-time data (grid, load and PV) would be taken over more often

Cosik, 2025-04-08:
@BastiBerlin93 I propose using OpenDTU then you could have resolution even 30 seconds or less. Could you show/prove that data is send very 5 min? I switch to OpenDTU for few years.

BastiBerlin93, 2025-04-08:
@Cosik 

[image]

Unfortunately, I cannot use Open DTU, which I operate both a DTU Pro and a DTS on a HAT. Attached is a screenshot of the times the update came with V5.02 of the DTU. The DTS at HAT can always do this including real-time data.

BastiBerlin93, 2025-04-08:
@Cosik 

Here is another video of the real-time data

https://github.com/user-attachments/assets/67b9ba5a-2f4d-48a1-a8a4-e0f8d2eeebf1

Wind06, 2025-04-25:
Hello, 

I confirm S-Miles cloud is now updated very quickly around every 10 seconds. could it be possible to adapt the addon ?
to be aligned with the cloud ? 
 
Thanks

Cosik, 2025-04-27:
I checked and uInverters are still updating per 15 min. I think it is related to energy meter. But still pooling data from cloud so often it is not recommended, or you want to make this addon banned?

BastiBerlin93, 2025-04-28:
I have a DTU Pro + Chint energy meter. The DTU has version 5.02 (first version for 5 minute update).

The real-time update only works with the hybrid inverters (HAT/HYT/...)

If 10 seconds is too fast, would it be necessary to make a constant query ? for example 1 minute to always have the current values. Currently it takes 5-10 minutes to update and the values always lag behind the CLoud

Wind06, 2025-04-28:
> I checked and uInverters are still updating per 15 min. I think it is related to energy meter. But still pooling data from cloud so often it is not recommended, or you want to make this addon banned?

Hi Cosik,
Please don't take this the wrong way — it was just a suggestion. Of course, I don't want to break anything.
I currently have two configurations: one with a DTU-Pro for solar microinverter reporting, which seems to update less frequently.
My second setup is an HAS-LV-EUG1 with batteries, and this one updates more frequently — every 10 seconds using the miles-cloud application.

Thanks for your support

Cosik, 2025-04-29:
@Wind06 and @BastiBerlin93 there is PR #197 ready for testing, please check if before I will merge it.

BastiBerlin93, 2025-04-29:
Is the update already available in HA?

Cosik, 2025-04-29:
You should use my repo for testing purpose.

BastiBerlin93, 2025-04-29:
How do I do that? Sorry but I have never done this before

Cosik, 2025-04-29:
the same way as you've done it with this repo.
Here is my fork used for developement https://github.com/Cosik/hoymiles/tree/master

BastiBerlin93, 2025-04-29:
I can't see the Edge version in the add-on store

BastiBerlin93, 2025-04-29:
I Can´t See Update to stable v1.4.4

BastiBerlin93, 2025-04-29:
works now get the current values every 5 minutes and do not lag behind

Cosik, 2025-04-30:
@Wind06 and @BastiBerlin93 are you happy with solution? Could be merged to official release?

closed 2025-05-01 by Cosik