dmslabsbr/hoymiles #190: Enities and values to use for Home Assistant Energy Consumption

wojciehm, 2025-04-02:
Hey folks. First of all thanks for the great work you are doing!

I am new to Home Assistant and perhaps my questions might be already answered or simply wrong. In HAAS I configured Hoymiles addon with MQQT. I receive a lot of information. 

I use 800W microinverter and this is my production of electricity.

[image]

I tried several inputs as the real solar energy production. When I use solarH_XXXXXX_total_eq the values in HAAS/ Energy are IMO wrong. It shows total energy produced today as this. 
[image]

IMO it would be better to compare current electricity usage from grid vs what is produced by solar panels.

I think (correct me if I am wrong) it would be better to use  **solarH_XXXXXXX_real_power**.

[image]

Unfortunately **solarH_XXXXXXX_real_power** cannot be selected in HAAS Energy section

[image]

I though as well to use solarH_XXXXXX_real_power or solarH_XXXXXX_real_power_measurement which reflects current energy production but then I have this error.

[image]

This is screenshot from the dashboard I built which reflects Hoymiles portal data.

[image]

[image]

Can you advise here?

Cheers, 
Wojciech

Cosik, 2025-04-08:
@wojciehm energy dashboard is using energy entities not power. You have to choose correct one

wojciehm, 2025-04-08:
@Cosik Which one would you suggest to use?

Cosik, 2025-04-08:
@wojciehm like I mention, all entities which provide energy instead of power. Names are legacy and wrong but you could use:
- `real_power_measurement`
- `real_power_total_increasing`
I propose using last one. In new edge I will try to fix it.

swatchpeter, 2025-04-16:
Hello,
In Home Assistant, I’m also unable to integrate the solar panel consumption into the Energy module.
The Energy module expects values that increase over time for the solar panels, similar to how an electricity meter reading behaves.
However, the variable solarH_xxxxxxx_real_power_total_increasing does not actually increment — it goes up and down throughout the day, just like the variable solarH_xxxxxxxx_real_power_measurement.
Do you have any idea why this might be?

swatchpeter, 2025-04-22:
> Hello, In Home Assistant, I’m also unable to integrate the solar panel consumption into the Energy module. The Energy module expects values that increase over time for the solar panels, similar to how an electricity meter reading behaves. However, the variable solarH_xxxxxxx_real_power_total_increasing does not actually increment — it goes up and down throughout the day, just like the variable solarH_xxxxxxxx_real_power_measurement. Do you have any idea why this might be? Thanks in advance
> 
> Real Power Measurement ==> OK [image]
> 
> Real Power Total Increasing ==> Not Increasing [image]

Hello,
I have create a workaround in order to use the Energy with my Solar panels : 
==> Create a Helpers Sensor to add to existing values
{{states('sensor.hoymiles_gateway_solarh_xxxxxxx_total_eq')|float+states('sensor.hoymiles_gateway_solarh_xxxxxxx_today_eq')|float}}

(replace xxxxxx by the ID in your Home Assistant system)

[image]

and it's working well !

closed 2025-04-29 by Cosik

swatchpeter, 2025-04-30:
Hello,
Regarding the definition of a cumulative consumption sensor, I was too hasty.
While the consumption is indeed cumulative, at midnight and around 5 a.m., the HoyMiles sensors update, which causes a dip in the result as shown in the graph below.
I will continue working on it to find a more reliable solution.
To be continued ....
[image]