dmslabsbr/hoymiles #191: Unable to connect to hoymiles server

kucinskimarek, 2025-04-08:
Hi.
Tonight the add-on stopped working. It is possible that it had to do with some updates? (a new look for the user page and app).
Login credentials unchanged, everything worked without a problem until yesterday.

`2025-04-08 06:27:00,097 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.4.1
2025-04-08 06:27:00,097 - HoymilesAdd-on - INFO - Starting up... 2025-04-08 06:27:00
2025-04-08 06:27:00,098 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2025-04-08 06:27:00,099 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2025-04-08 06:27:00,099 - HoymilesAdd-on.mqttapi.Mqtt - INFO - mqtt.Client b6c7725a-1431-11f0-aa6d-86baf935f6f0
2025-04-08 06:27:00,099 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT core-mosquitto
2025-04-08 06:27:00,104 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2025-04-08 06:27:00,105 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to core-mosquitto
2025-04-08 06:27:01,106 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2025-04-08 06:27:02,472 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Access error: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2025-04-08 06:27:02,472 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Status code: 404Not Found
2025-04-08 06:27:02,473 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Wrong user/password 404 Not Found
2025-04-08 06:27:02,474 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - I can't get access token`

Restarting HA nor the add-on did not change anything.
Any ideas?

Cosik, 2025-04-08:
Hi @kucinskimarek and you check if page is reachable?

kucinskimarek, 2025-04-08:
> Hi [@kucinskimarek](https://github.com/kucinskimarek) and you check if page is reachable?

Yes, the page and HM app works ok and presents data.

Cosik, 2025-04-08:
Nope, you have authentication page in log which return 404 few minutes ago

hangingloose, 2025-04-08:
same problem... 
It looks like Hoymiles change something on their Homepage... 
They have a new design 👎

kucinskimarek, 2025-04-08:
> Nope, you have authentication page in log which return 404 few minutes ago

I thought you meant the client webside, where we can log in and check data. 
https://global.hoymiles.com/website/

hangingloose, 2025-04-08:
The page https://global.hoymiles.com/platform/api/gateway/iam/auth_login
give me now the result "404 Not Found"
If I want to get the access to the hoymiles I use 
https://global.hoymiles.com/website/login
https://global.hoymiles.com/website/home
my shortlink in my browser has:
https://global.hoymiles.com/platform/station/view

it looks like they Change from "platform" to "website"

Cosik, 2025-04-08:
You are referring to web URL not API used in add-on. Without proper investigation it will not help

storkio, 2025-04-08:
Same here since today

`2025-04-08 07:22:35,171 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.4.1
2025-04-08 07:22:35,171 - HoymilesAdd-on - INFO - Starting up... 2025-04-08 07:22:35
2025-04-08 07:22:35,172 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2025-04-08 07:22:35,172 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2025-04-08 07:22:35,172 - HoymilesAdd-on.mqttapi.Mqtt - INFO - mqtt.Client 7aa357c8-1439-11f0-acb5-d83add322995
2025-04-08 07:22:35,172 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT core-mosquitto
2025-04-08 07:22:35,176 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2025-04-08 07:22:35,176 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to core-mosquitto
2025-04-08 07:22:36,177 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2025-04-08 07:22:36,585 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Access error: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2025-04-08 07:22:36,586 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Status code: 404Not Found
2025-04-08 07:22:36,589 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Wrong user/password 404 Not Found
2025-04-08 07:22:36,590 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - I can't get access token
2025-04-08 07:28:42,390 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - I can't get access token `

hangingloose, 2025-04-08:
> You are referring to web URL not API used in add-on. Without proper investigation it will not help

Correct! But could it be, the the API could be the also on "webside" now? 
Do you have access to a hoymiles plant?

kucinskimarek, 2025-04-08:
> You are referring to web URL not API used in add-on. Without proper investigation it will not help

I am not a programmer and I am asking a bit out of curiosity. How is such data obtained? For example, does HM Support willingly cooperate with such inquiries, or do you have to investigate on your own? Can we help somehow? 
Maybe someone with installer credentials has access to more information?

kucinskimarek, 2025-04-08:
> > You are referring to web URL not API used in add-on. Without proper investigation it will not help
> 
> Correct! But could it be, the the API could be the also on "webside" now? Do you have access to a hoymiles plant?

I tried randomly several combinations with word substitutions end link modifications in previous API link , but with no results. Still 404 error, but already on the new page layout :)

sirteemu, 2025-04-08:
Same problem here, no longer works.

Gielz1986, 2025-04-08:
No longer working here, anyone able to test it with the following url?

https://previous.hoymiles.com/platform/api/gateway/iam/auth_login

Rezessivo, 2025-04-08:
same here with another Trace-ID

PouletZlika, 2025-04-08:
Hoymiles login page seems to have changed during the night :-( I think we have to wait for a new version of the add-on.
Doas anyon knows if the authentication URL is a parameter in some config file  ?

kneissel, 2025-04-08:
> No longer working here, anyone able to test it with the following url? I'm currently at work and unable to figure out where to change it.
> 
> https://previous.hoymiles.com/platform/api/gateway/iam/auth_login

I've made a test with previous.hoymiles.com and it works, could be a quick temporary fix.

`dmslabs - Home Assistant HoyMiles Solar Data Gateway Add-on
2025-04-08 09:20:04,944 - HoymilesAdd-on - INFO - ********** dmslabs&Cosik Hoymiles Gateway  v.1.4.1
2025-04-08 09:20:04,944 - HoymilesAdd-on - INFO - Starting up... 2025-04-08 09:20:04
2025-04-08 09:20:04,945 - HoymilesAdd-on - INFO - Using Internal MQTT Server: core-mosquitto
2025-04-08 09:20:04,945 - HoymilesAdd-on - INFO - Using Internal MQTT User: addons
2025-04-08 09:20:04,945 - HoymilesAdd-on.mqttapi.Mqtt - INFO - mqtt.Client e4a15de0-1449-11f0-a04e-bc241158a783
2025-04-08 09:20:04,945 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Starting MQTT core-mosquitto
2025-04-08 09:20:05,165 - HoymilesAdd-on.mqttapi.Mqtt - INFO - MQTT connected with result code 0
2025-04-08 09:20:05,166 - HoymilesAdd-on.mqttapi.Mqtt - INFO - Connected to core-mosquitto
2025-04-08 09:20:05,951 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/iam/auth_login
2025-04-08 09:20:08,861 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2025-04-08 09:20:08,861 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm/station_find
2025-04-08 09:20:09,520 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2025-04-08 09:20:10,228 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:10,900 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:11,592 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:12,305 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:12,633 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:13,291 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:14,005 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:14,876 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2025-04-08 09:20:15,696 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2025-04-08 09:20:16,066 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2025-04-08T09:10:02+02:00
2025-04-08 09:20:16,066 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:16,436 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:17,104 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:17,917 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:18,629 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:18,969 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:19,613 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-04-08 09:20:20,275 - HoymilesAdd-on - INFO - Solar data publication...2025-04-08 09:20:20.275802
2025-04-08 09:20:20,278 - HoymilesAdd-on - INFO - DTU-Pro-S_4559080 data publication...2025-04-08 09:20:20.278346
2025-04-08 09:20:20,281 - HoymilesAdd-on - INFO - HMS-1000-2T_14888670 data publication...2025-04-08 09:20:20.281357
2025-04-08 09:20:20,286 - HoymilesAdd-on - INFO - HMS-1000-2T_14888630 data publication...2025-04-08 09:20:20.286046
2025-04-08 09:20:20,289 - HoymilesAdd-on - INFO - HMS-1000-2T_14888620 data publication...2025-04-08 09:20:20.289241
2025-04-08 09:20:20,295 - HoymilesAdd-on - INFO - HMS-1000-2T_14888660 data publication...2025-04-08 09:20:20.295927
2025-04-08 09:20:20,309 - HoymilesAdd-on - INFO - HMS-1000-2T_14888640 data publication...2025-04-08 09:20:20.309888
2025-04-08 09:20:20,311 - HoymilesAdd-on - INFO - HMS-1000-2T_14888650 data publication...2025-04-08 09:20:20.310969
2025-04-08 09:20:20,325 - HoymilesAdd-on - INFO - HMS-1000-2T_14888680 data publication...2025-04-08 09:20:20.325735
2025-04-08 09:20:20,346 - HoymilesAdd-on - INFO - Main loop start!
 Live`

hangingloose, 2025-04-08:
where can I find this files?

Ahgriel, 2025-04-08:
Bonjour .
Comment peut on modifier ces liens ?

kneissel, 2025-04-08:
I just make a pull request with changes (sorry but i'm not really "fluent" with github so maybe i've not done what is expected to be done.)

kneissel, 2025-04-08:
> where can I find this files?

It can't be done directly in Home Assistant, you should do it directly int this github repository or make a fork one and use it in Home Assistant (it's better to wait for change here as using a fork could be a bad idea if you've not plan to keep it update)
And please, don't use my fork, i'll delete it as soon as it'll be fixed here.

Kwiatjk, 2025-04-08:
Same problem 
2025-04-08 11:40:06,217 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2025-04-08 11:40:07,841 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Access error: https://global.hoymiles.com/platform/api/gateway/iam/auth_login
2025-04-08 11:40:07,842 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Status code: 404Not Found
2025-04-08 11:40:07,844 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - Wrong user/password 404 Not Found
2025-04-08 11:40:07,844 - HoymilesAdd-on.hoymilesapi.Hoymiles - ERROR - I can't get access token

Cosik, 2025-04-08:
I want to ask user who could add only comment like "same problem" to stop spamming. I fast check two PRs and they are masking problem, not solving them. I will have time in the evening to work on it.

bl4ckfir3, 2025-04-08:
I installed it and it dont work for me.

[image]

kucinskimarek, 2025-04-08:
Yep. Something not quite right.

`Traceback (most recent call last):
  File "/hoymiles.py", line 31, in <module>
    from hoymilesapi import Hoymiles
  File "/hoymilesapi.py", line 18, in <module>
    from const import (
ImportError: cannot import name 'BASE_URL' from 'const' (/const.py)
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

Cosik, 2025-04-08:
yes, done in rush, now should work

NDBassie, 2025-04-08:
Thanks @Cosik for the update! Unfortunately running into the same issue as  kucinskimarek

kucinskimarek, 2025-04-08:
Yep. 1.4.3 works like a charm!
Thanks for the quick fix.

bl4ckfir3, 2025-04-08:
Confirm it work now

Cosik, 2025-04-08:
Just check, that new url is `https://neapi.hoymiles.com/iam/pub/0/auth/login` so I will try to check it during night.

PouletZlika, 2025-04-08:
Thanks a lot for the temporary fix ! Great job !

savaskapsis, 2025-04-08:
Good evening, sorry for the silly question, but how do I update? I have the 14.1 version.

bl4ckfir3, 2025-04-08:
> Good evening, sorry for the silly question, but how do I update? I have the 14.1 version.

Search for Updates in Home Assistenten and them appears the Update to 1.4.3 Version :-)

[image]

savaskapsis, 2025-04-08:
Thank you very much, I was searching for it through the addons and couldn't find it.

abpei, 2025-04-08:
You da man!!

BastiBerlin93, 2025-04-09:
Worked yesterday after the update. Tonight again then the same problem

kucinskimarek, 2025-04-09:
> Worked yesterday after the update. Tonight again then the same problem

Strange, at my place it continues to work correctly.
Have you tried restart add-on? In the logs the same as yesterday?

Cosik, 2025-04-09:
@BastiBerlin93 logs...

BastiBerlin93, 2025-04-09:
Sorry, just looked and had other problems. It was just a coincidence that it was at the same time. Everything is running

I wanted to write it but you were faster

elfeo31, 2025-04-10:
Working again ! Thank you very much !

eikearends, 2025-04-11:
Issue was fixed with Version 1.4.3
Got another Problem now… can see on screenshot

[image]

tbd56, 2025-04-13:
> Issue was fixed with Version 1.4.3 Got another Problem now… can see on screenshot
> 
> Anyone got the same issue?
> 
> [image]

I can confirm the same!

Cosik, 2025-04-13:
Orginal issue is somehow solved. If there is any other please create new ticket will full logs. Debug one will be the best.

closed 2025-04-13 by Cosik