dmslabsbr/hoymiles #198: Power management (question)

ErikSabel, 2025-05-10:
In the Hoymiles app, there is an option to say on how much percentage the solar convertors may run.
E.g. normally the output is maximum 5 kWh. If you say 50% in that option, it will only generate 2,5 kWh

The reason for the request/question on the possibility is that we have for example tomorrow a negative power rate. For ever 1 kWh you return, you will need to pay 28 cents. (Other way around, if you consume 1 kWh you will get 28 cents for that)
So it is very interesting to turn off the solar panels on those moments via an automation

Is this option available in the current version that is running on home assistant? I couldn't find it, but perhaps someone else found it, or has a good idea how to use such a function

Eastmasters, 2025-05-18:
I’m interested in this functionality as well. Tibber is planning to switch to 15 minutes price blocks for dynamic pricing in the Netherlands. With the negative prices, there is a huge incentive to have automated control of power output.

ErikSabel, 2025-05-18:
> I’m interested in this functionality as well. Tibber is planning to switch to 15 minutes price blocks for dynamic pricing in the Netherlands. With the negative prices, there is a huge incentive to have automated control of power output.

The 15 minuten is postponed, not only tibber but it's in general. It isn't possible via this add on, I have solved it this week with an esp21 board plus an CMT. Running on opendtu
Data that you will get is more accurate than this app and pushed every 10 to 15 seconds

Eastmasters, 2025-06-13:
I found this integration with power adjustment support.  I haven't figured out how to add this repository yet. https://github.com/wil-lem/ha-hoymiles-s-cloud