dmslabsbr/hoymiles #199: Entity is currently unavailable

osmaterdca, 2025-05-15:
Hello.
This addon works great (worked great for many month), thanks a lot for your work !

However I'm experiencing an issue every day which force me to restart HA and wait for a variable period of time before it works again.

[image]

I tried to remove the add-on, redo the config but no luck.

Let me know which files/log/configuration you may need to help me to resolve this issue.

Thank you.
David

denisfln, 2025-06-12:
Same here. I think it's probably a connection issue after an MQTT update.

BarryH120, 2025-06-22:
> Change default configuration worked for me:
> 
> [image]

Where do you find this change default configuration?
thx