dmslabsbr/hoymiles #201: KeyError: 'reflux_station_data'

PouletZlika, 2025-07-17:
Hi, since 2 hours, the add-on has stopped and shows error every time I relaunch it (even after restarting HA) ;
Does anyone has the same issue or is it on my side ? (no modification at all on my HA since a while)

Here is the log

Thanks

025-07-17 13:22:42,515 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/iam/auth_login
2025-07-17 13:22:43,176 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - I got the token!!  :-)
2025-07-17 13:22:43,176 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm/station_find
2025-07-17 13:22:43,422 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2025-07-17 13:22:43,981 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-07-17 13:22:44,699 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-07-17 13:22:45,298 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-07-17 13:22:46,011 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-07-17 13:22:46,259 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-07-17 13:22:46,867 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-07-17 13:22:47,148 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-07-17 13:22:47,755 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_find_details
2025-07-17 13:22:48,403 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm/station_select_device_of_tree
2025-07-17 13:22:49,183 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - Loading: https://previous.hoymiles.com/platform/api/gateway/pvm-data/data_count_station_real_data
2025-07-17 13:22:49,948 - HoymilesAdd-on.hoymilesapi.Hoymiles - INFO - last_data_time 2025-07-17T13:12:26+02:00
Traceback (most recent call last):
  File "/hoymiles.py", line 507, in <module>
    sys.exit(main())  # next section explains the use of sys.exit
  File "/hoymiles.py", line 462, in main
    publicate_data(plant_list[id], mqtt_h)
  File "/hoymiles.py", line 263, in publicate_data
    hoymiles_h.get_solar_data()
  File "/hoymilesapi.py", line 268, in get_solar_data
    self.solar_data = self.adjust_solar_data(self.solar_data)
  File "/hoymilesapi.py", line 344, in adjust_solar_data
    del solar_data["reflux_station_data"]
KeyError: 'reflux_station_data'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

PouletZlika, 2025-07-17:
So it's probably on Hoymiles side. Thanks guys.

claudealain59, 2025-07-17:
Same problem. Toke 3 hours to search ... Perplexity says : it's github side ... I reinstalled 5 times lol
Mayday Maydays ! :-)

PouletZlika, 2025-07-17:
Probably the authentication url previous.hoymiles… has been disabled by Hoymiles 

We Just have to let the devs of this add on the time to analyze the issue and eventually have a solution.

It’s a free add on they develop on their free time so … we just have to be patient

savaskapsis, 2025-07-17:
Thank you very much for resolving this so quickly

PR #204 (2025-07-17): Implemented fix for missing reflux data
As suggested in #203

This also fixes #201 and #203