dmslabsbr/hoymiles #203: ## ❗ Bug report: KeyError on missing "reflux_station_data"

gregi84, 2025-07-17:
Hi 👋

Thanks for your great add-on. It worked perfectly until recently.

Since 1 to 3 days, the add-on crashes with the following error:

```
KeyError: 'reflux_station_data'
```

The error seems to come from this line in `hoymilesapi.py` (function `adjust_solar_data`):

```python
del solar_data["reflux_station_data"]
```

However, this key seems not to be always present anymore in the API response from Hoymiles.

### ✅ Suggested fix:

Please replace it with the safer version below to avoid a crash:

```python
if "reflux_station_data" in solar_data:
    del solar_data["reflux_station_data"]
```

This would prevent the plugin from crashing when the key is absent (e.g., when `is_reflux = 0`).

I'm using Home Assistant OS and cannot modify the add-on files directly, so an update would be very helpful.

Thanks again for your great work 🙏  
Greg

gjabma, 2025-07-17:
or :  solar_data.pop("reflux_station_data", None)

Krydo1337, 2025-07-17:
> please update and retest with v1.4.5

That, was fast... Thank you for updating. It works :)

Ahgriel, 2025-07-18:
Comment obtenir la version 1.4.5  ?
Je ne vois rien pour forcer la MAJ
Merci.

closed 2025-08-08 by Cosik