dmslabsbr/hoymiles #212: Two separate DTUs in one PV installation - how to connect to HA?

Krzysztonek, 2025-09-26:
Hi, @Cosik and @dmslabsbr 

I've been using one DTU-Mi model so far, now I need to extend my solar plant adding some new panels.
These new panels will be connected to a separate DTU-WLite S, since old and new microinverters are not compatible with each other. Not possible to have all the panels connected to the same DTU. 
Both DTUs will be registered and assigned to the same Hoymiles account (same Hoymiles User, same Hoymiles plant ID). From Hoymiles perspective it will be ONE plant, regardless the number of DTUs. 

Do I need to add a new Hoymiles add-on instance to my HA to collect data from DTU-WLite S? 
Will they be treated as a common or sepatate data collection?

Cosik, 2025-09-26:
@Krzysztonek just check it.

Krzysztonek, 2025-09-28:
> Do I need to add a new Hoymiles add-on instance to my HA to collect data from DTU-WLite S?

### **No, I dont!**

Regardless the DTU model all the devices registered at the same user account are treated as a common data source, providing already aggregated data to HA instance.

The only drawback of this solution is that for each DTU I had to write a separate script that collects production data from individual panels connected to different DTU models.